ros/geometry2 #358: tf2_eigen: Should Eigen::Affine3d be converted to tf2::Transform?

peci1, 2019-02-11:
Affine transforms are a superset of ROS transforms (which are isometries). So should there really be the conversion function from Affine3d to Transform? Shouldn't the user be required to convert the Affine3d to Isometry3d in Eigen first (doing all the needed checks)? Or should the tf2 conversion function issue a warning/error when the given affine transform is not an isometry?

tfoote, 2019-02-14:
It would probably be good to deprecate that part of the API. It makes more sense since it's not an isomorphic conversion.  When first written the Affine3d was the only transform available and Isometry3d was added later and support added here recently (#335). Though we will want to target that for the next distro to not break people in the current distro.

peci1, 2019-02-15:
Do you think a build time deprecation/warning or rather a runtime warning would be suitable?

It might be nice to actually check if the given transform is an isometry or not, and issue a runtime warning when it isn't, but I can imagine that could have impact on performance. And in most cases the transform will be an isometry (because it probably comes from a tf transform).

tfoote, 2019-02-15:
A build time deprecation warning is appropriate. This can only be fixed by changing code. Spamming the user will just be annoying.

peci1, 2019-03-05:
Deprecation warnings added in #371 .

peci1, 2024-02-03:
Noetic continuation in #561.

closed 2025-05-31 by sloretz