ros/geometry2 #359: Unused parameters in message_filter.h

Tuebel, 2019-02-18:
When compiling my node which depends on the tf2_ros message filter I get the following warning.

```make
/home/tim/robo_guide_ws/src/robo_guide/src/nodelets/tracker_nodelet.cpp:56:50:   required from here
/opt/ros/melodic/include/tf2_ros/message_filter.h:455:89: warning: unused parameter ‘target_frame’ [-Wunused-parameter]
   void transformable(tf2::TransformableRequestHandle request_handle, const std::string& target_frame, const std::string& source_frame,
                                                                                         ^~~~~~~~~~~~
/opt/ros/melodic/include/tf2_ros/message_filter.h:455:122: warning: unused parameter ‘source_frame’ [-Wunused-parameter]
   void transformable(tf2::TransformableRequestHandle request_handle, const std::string& target_frame, const std::string& source_frame,
                                                                                                                          ^~~~~~~~~~~~
/opt/ros/melodic/include/tf2_ros/message_filter.h:456:32: warning: unused parameter ‘time’ [-Wunused-parameter]
                      ros::Time time, tf2::TransformableResult result)
                                ^~~~
```

As the message_fitler works as expected I assume that the parameters of the function can be deleted to avoid the warning.

tfoote, 2019-02-20:
Yes, the extra parameters are unused at the moment. I believe that they are there in case the information could be used to optimize the algorithm to only query transformability for frames related to the changed frames instead of checking all. However checking all was made more efficient and the usefulness of those arguments is lower.

Tuebel, 2019-02-21:
Thanks for your answer.
So I do understand it right, that they could be removed but remain in the code for API stability?

tfoote, 2019-02-26:
Correct

closed 2019-03-04 by Tuebel