ros/geometry2 #381: canTransform() with timeout while unloading nodelet with paused time -> freeze

peci1, 2019-05-16:
I have a nodelet which periodically calls `canTransform()` with timeout in a separate thread on a TF2 ROS buffer.

When I'm replaying a bag file and pause it, the `canTransform()` call (correctly) hangs until I unpause the playback. However, if I want to unload the nodelet while the bag is paused, it never unloads because the `canTransform()` call is still hanged.

For nodes, this was already solved in https://github.com/ros/geometry2/issues/26 . However, in nodelet case, there is nothing like `ros:ok()` to check for shutdown requests.

What can I do to mitigate this behavior? 

Until this is resolved, I think I'll write my own helper `Nodelet::canTransform()` doing the waiting loop itself and checking for some unload/stop flag.

tfoote, 2019-05-17:
I don't have a good idea of how to resolve this we don't have a good signaling mechanism to say when to stop waiting. I think that your custom wrapper which interjects the additional check for a nodelet shutdown flag is likely going to be the best approach in the short term.

Other approaches might be to force a time update on unload, but I don't think that you can do that without effecting other parts of the system. But a clock publisher that jumped time briefly so that the timeout would clear would work to allow arbitrary nodelets to shutdown without code changes.

peci1, 2020-06-28:
I've created a mixin called `StatefulNodelet`:

```c++
class StatefulNodelet
{
public:
  virtual ~StatefulNodelet() { this->shutdown(); }

  /**
   * \brief Similar to ros::ok(). Becomes false when nodelet unload is requested.
   * \return Whether it is ok to continue nodelet operations.
   */
  bool ok() const { return !this->isUnloading; }

protected:
  void shutdown() { this->isUnloading = true; }

private:
  bool isUnloading = false;
};
```

If the user adds this mixin to his/her nodelet implementations, it is then possible to use an adapted TF2 buffer which checks the `ok()` flag of the nodelet. Do you think it's worth sending a PR towards the nodelet package that would make this mixin publicly available? That would allow to create the properly working TF2 buffer version here.

peci1, 2022-02-04:
PR in nodelet_core created: https://github.com/ros/nodelet_core/pull/116

closed 2025-05-31 by sloretz