ros/geometry2 #385: Error in rosbag play (Transform[sender=unknown_publishser]

Ashik19, 2019-06-06:
Dear Altruist people,

When I am trying to play the pointcloud2 topic in the legloam of a pre-recorded bag file with "map" as the fixed frame I am getting an error

[image]

The herewith following picture is the launch file 

[image]

The rqt_tf_tree is as the following

[image]

When I am running rosrun rqt_tf_tree rqt_tf_tree, I am also getting a warning like the following picture-

[image]

Can you please help me what should I do to resolve this issue ?

Thank you very much for your valuable time.

Thanks 
Ashik

tfoote, 2019-06-06:
Please ask questions on http://answers.ros.org according to our support policy: http://wiki.ros.org/Support We try to keep the issue tracker reserved for bugs and feature developments.

closed 2019-06-06 by tfoote

migiuliani, 2023-06-29:
Hey guys,
were you able to solve this issue? I have the same problem trying to run Autoware-Carla simulation environment.
Thank you in advance