ros/geometry2 #387: tf2::doTransform<sensor_msgs::PointCloud2> doesn't transform normals and viewpoints

peci1, 2019-07-01:
I've noticed that tf2::doTransform<sensor_msgs::PointCloud2> from tf2_sensor_msgs only transforms the XYZ fields, and leaves normals and viewpoints untransformed. I can either send a PR adding support for this (if the channels are present), or there should be a warning that the pointcloud was transformed but not its normals and viewpoints. What solution does sound better?

There could also be a new helper function called `transformChannel` or similar, that would take a pointcloud, channel prefix and transform, and apply the transform to all points in the given channel (appending `x`, `y` and `z` to the channel prefix).

tfoote, 2019-07-10:
A PR to extend support for the other channels would be great.

peci1, 2019-07-10:
See #396.

closed 2025-05-31 by sloretz