ros/geometry2 #389: [tf_eigen] Intended use of toMsg specialization

artivis, 2019-07-08:
I wonder what is the intended use of the following conversion specialization ([tf2_eigen.h#L154](https://github.com/ros/geometry2/blob/melodic-devel/tf2_eigen/include/tf2_eigen/tf2_eigen.h#L154)) introduced in #294, 
```cpp
geometry_msgs::Vector3& toMsg(const Eigen::Vector3d& in, geometry_msgs::Vector3& out)
````
since it does not seems to follow the specialization pattern of `toMsg`.
Is it meant to be called directly rather than through `tf2::convert` ??

Adding the following test snippet to `tf2_eigen`'s tests results in a compilation error:
```cpp
TEST(TfEigen, ConvertVector3dToVector3)
{
  const Eigen::Vector3d v(1,2,3);

  Eigen::Vector3d v1;
  geometry_msgs::Vector3 p1;
  tf2::convert(v, p1);
  tf2::convert(p1, v1);

  EXPECT_EQ(v, v1);
}
```

Error:
```terminal
geometry2/tf2/include/tf2/impl/convert.h: In instantiation of ‘static void tf2::impl::Converter<IS_MESSAGE_A, IS_MESSAGE_B>::convert(const A&, B&) [with A = Eigen::Matrix<double, 3, 1>; B = geometry_msgs::Vector3_<std::allocator<void> >; bool IS_MESSAGE_A = false; bool IS_MESSAGE_B = true]’:
geometry2/tf2/include/tf2/convert.h:51:113:   required from ‘void tf2::convert(const A&, B&) [with A = Eigen::Matrix<double, 3, 1>; B = geometry_msgs::Vector3_<std::allocator<void> >]’
geometry2/tf2_eigen/test/tf2_eigen-test.cpp:65:21:   required from here
geometry2/tf2/include/tf2/impl/convert.h:65:5: error: no match for ‘operator=’ (operand types are ‘geometry_msgs::Vector3_<std::allocator<void> >’ and ‘geometry_msgs::Point {aka geometry_msgs::Point_<std::allocator<void> >}’)
   b = toMsg(a);
   ~~^~~~~~~~~~
In file included from /opt/ros/melodic/include/geometry_msgs/Transform.h:18:0,
                 from /opt/ros/melodic/include/geometry_msgs/TransformStamped.h:19,
                 from geometry2/tf2/include/tf2/transform_functions.h:37,
                 from geometry2/tf2_eigen/include/tf2_eigen/tf2_eigen.h:32,
                 from geometry2/tf2_eigen/test/tf2_eigen-test.cpp:30:
/opt/ros/melodic/include/geometry_msgs/Vector3.h:22:8: note: candidate: constexpr geometry_msgs::Vector3_<std::allocator<void> >& geometry_msgs::Vector3_<std::allocator<void> >::operator=(const geometry_msgs::Vector3_<std::allocator<void> >&)
 struct Vector3_
        ^~~~~~~~
/opt/ros/melodic/include/geometry_msgs/Vector3.h:22:8: note:   no known conversion for argument 1 from ‘geometry_msgs::Point {aka geometry_msgs::Point_<std::allocator<void> >}’ to ‘const geometry_msgs::Vector3_<std::allocator<void> >&’
/opt/ros/melodic/include/geometry_msgs/Vector3.h:22:8: note: candidate: constexpr geometry_msgs::Vector3_<std::allocator<void> >& geometry_msgs::Vector3_<std::allocator<void> >::operator=(geometry_msgs::Vector3_<std::allocator<void> >&&)
/opt/ros/melodic/include/geometry_msgs/Vector3.h:22:8: note:   no known conversion for argument 1 from ‘geometry_msgs::Point {aka geometry_msgs::Point_<std::allocator<void> >}’ to ‘geometry_msgs::Vector3_<std::allocator<void> >&&’
```

Clearly the 'proper specialization' is not found.

IanTheEngineer, 2019-07-09:
Good find and test @artivis . I need to go back through my notes on this, but I believe your assessment is correct. I needed this function to exist to do some conversions for [MoveIt to fully convert to `tf2`](https://github.com/ros-planning/moveit/pull/830), using the `toMsg` or `fromMsg` directly where appropriate, and sometimes using the `convert` methods. I don't recall if I used the `convert` method directly for these functions.

artivis, 2019-07-10:
Thanks for the reply. This issue is unfortunate as it gives the false feeling that `convert` would work for those types too. Looking at the `convert` traits mechanism there doesn't seems to be a simple fix without breaking something.

closed 2025-05-31 by sloretz