ros/geometry2 #403: tf2_ros::MessageFilter alternative for Python

hdamron17, 2019-08-15:
I am constantly rewriting the code to subscribe to a Pose or PointCloud topic then transform to the frame I intend to work in within the callback. In C++, I can use a message filter to achieve this, but there is no equivalent for Python. Is this intentionally left out or is it just not yet implemented? If the latter is the case, I would be glad to help implement it. I have not contributed to tf2 before so I may need some guidance with conforming to the current standards, but I am comfortable with Python, tf2, and message_filters.

tfoote, 2019-08-16:
Yeah, it's just not implemented yet. A contribution would be appreciated. Please look at adding it to tf2_ros and provide a parallel API to the c++ version.

hdamron17, 2019-08-20:
Looking at the C++ implementation, can you explain what is the purpose of using requests for the transformations? ([see here](https://github.com/ros/geometry2/blob/melodic-devel/tf2_ros/include/tf2_ros/message_filter.h#L331)) Is this just how it must be done in C++, or does this improve efficiency by allowing multiple requests to run in parallel and allowing messages to be dropped if the transformations take too long? I was planning to take a much simpler approach of just running each message through the [BufferInterface.transform](https://github.com/ros/geometry2/blob/melodic-devel/tf2_ros/src/tf2_ros/buffer_interface.py#L49) method, but let me know if I should follow the same pattern as the C++ implementation.

hdamron17, 2019-08-21:
I see now that I was misunderstanding the message filter. It does not actually transform the message but rather ensures that the message could be transformed. This also answers my question about what happens in the case of multiple target frames. Unfortunately, the ability of BufferCore to perform a callback whenever a transformation becomes available did not make it into the tf2_py API, probably because it is marked for internal use only. Before I dive into adding this to the API, do you know if there are any alternatives? @tfoote

closed 2025-05-31 by sloretz