ros/geometry2 #409: Duration parameter of tf2_ros::Buffer has no effect ?

nyxrobotics, 2019-08-28:
I wrote my code below.
> tf2_ros::Buffer(ros::Duration(0.3), true)
Then, I checked if it has applied.
> ros::Duration duration_tmp = tfBuffer.getCacheLength();
> ROS_INFO("cache time [%f]s",duration_tmp.toSec());
That shows the parameter is set to 0.3. However, every member functions of tfBuffer works with every TFs if they are very old data. 
For example, tfBuffer.getFrameStrings increases the size infinitely.

tfoote, 2019-09-11:
Please provide a reproducible example or extend one of the unit tests to show the problem.

Are you referencing static transforms? They are always valid and the cache duration does not apply to them. 

As an aside 0.3 seconds for the cache is very short, if there is any noteable network delays you're likely to run into challenges getting a spanning set of transforms.

tfoote, 2019-10-15:
I think that you may be seeing that the cache length is per transform and as such you may have 0.3 seconds of content from a long time ago if a transform has been stopped publishing. 

A PR for a method to iterate an prune old frames would be considered. We don't do that on receipt as that is the most exercised code path and consequently checks there would be expensive. To make it easier to use a helper class could be created in tf2_ros similar to the Listener that takes a buffer as an argument that would call the prune method with selected parameters on a timer if this is causing an issue.

closed 2025-05-31 by sloretz