ros/geometry2 #410: Feature request: Manual tensorflow manipulation

rickstaa, 2019-09-04:
First of all, thanks for the amazing tool it really helps me in my robotics projects! There is, however, one useful feature that I'm missing. After needing to manually change my tf2 frames, I stumbled upon this very useful repository [https://github.com/PickNikRobotics/tf_visual_tools](https://github.com/PickNikRobotics/tf_visual_tools) by [@PickNikRobotics](https://github.com/PickNikRobotics. I think incorporating this function in the main package would be very useful.

tfoote, 2019-09-12:
With the high level dependencies that package relies upon it's not suitable to be merged into the core package. For example it has gui requirements that are not approriate for a low level package. However a contributed tutorial to make the package more visible would be helpful for other users.

closed 2019-09-12 by tfoote

rickstaa, 2019-09-30:
@tfoote Thanks for your response. Good idea, I will make a contributed tutorial if I can get the package to work on melodic [see issue](https://github.com/PickNikRobotics/tf_visual_tools/issues/6).

rickstaa, 2019-10-18:
**Update:** As I could not fix the problems I had with the package under ROS melodic I am reluctant to write a contributed tutorial at the moment. I, however, opened an issue might the original authors/maintainers of the package [/tf_visual_tools/issues/7](https://github.com/PickNikRobotics/tf_visual_tools/issues/7) want to write one.

artemiialessandrini, 2020-04-02:
@rickstaa Good, tool works in melodic now.
[ /tf_visual_tools/issues/6](https://github.com/PickNikRobotics/tf_visual_tools/issues/6)