ros/geometry2 #411: geometry2 / tf2_py incompatible with python 3.6 and 3.7

Simon-Steinmann, 2019-09-14:
Just in case someone runs into it with ROS Melodic on Ubuntu 18.04, because presented solution which worked fine on 16.04 does not work anymore. I managed to get around it with:

`catkin_make -DPYTHON_EXECUTABLE:FILEPATH=/path/to/virtualenv/python`

or in case that you are using `catkin build`:

`catkin build --cmake-args -DPYTHON_EXECUTABLE:FILEPATH=/path/to/virtualenv/python`

_Originally posted by @Jendker in https://github.com/ros/geometry2/issues/259#issuecomment-513596006_

So I painstakingly figured out that while yes, this fix for 18.04 works, however make sure you create a virtual environment for python 3.5, python 3.6 and 3.7 DO NOT WORK

Jendker, 2019-09-15:
Well it works for me with Python 3.6 installed on Ubuntu 18.04 with apt.

How do you select Python version?

Simon-Steinmann, 2019-09-16:
Well the compiling part works with python 3.6 and 3.7. However, importing the package after, the error persists. Only compiling with 3.5 worked for me. 

You can set up virtual environments with the python of your choice. I use Anaconda.

Jendker, 2019-09-16:
I am using Python 3.6 and can import `tf2_ros` or `tf2_py` without any problems. Maybe it's somehow related to Anaconda?

What about just using Python 3.6 delivered with apt? It works for me without any problems.

`sudo apt install python3`

Simon-Steinmann, 2019-09-16:
I'll try again

Simon-Steinmann, 2019-09-16:
I stand corrected, tried it with 3.7.4 on a fresh install and it worked. I apologize for the confusion

Jendker, 2019-09-16:
I'm glad you made it work 👍

If everything is fine you can close the issue :)

closed 2019-09-16 by Simon-Steinmann