ros/geometry2 #412: using rosbag API to copy tf_static

phil0stine, 2019-10-01:
I would like to fix a bag that's already been taken and that has more than one `tf_static` publisher, using the rosbag api. Here's a min working example of my current issue (I assume it's related to the connection headers for the `tf_static` topic).

* Publish tf_static
```
$ rosrun tf2_ros static_transform_publisher 0 0 0 0 0 0 A B
$ rosrun tf2_ros static_transform_publisher 1 1 1 1 1 1 A C
```
* Record a bag
```
$ rosbag record -a -O mwe.bag
```
* Play it back, and subscribe to `/tf_static` in another terminal (only the second transform is returned, due to `tf_static` latching)
```
$ rosbag play mwe.bag -k
$ rostopic echo /tf_static
transforms:
  -
    header:
      seq: 0
      stamp:
        secs: 1569941384
        nsecs: 924582898
      frame_id: "A"
    child_frame_id: "C"
    transform:
      translation:
        x: 1.0
        y: 1.0
        z: 1.0
      rotation:
        x: 0.167518791246
        y: 0.570941471358
        z: 0.167518791246
        w: 0.786066629137
---
```
* Use rosbag API to simply do a passthrough copy to another bag
```
$ ipython
In [1]: import rosbag
In [2]: b=rosbag.Bag('mwe.bag')
In [3]: b_out=rosbag.Bag('mwe_out.bag', 'w')
In [4]: for topic, msg, t in b.read_messages():
   ...:     b_out.write(topic, msg, t)
   ...:
In [5]: b_topic_info = b.get_type_and_topic_info("/tf_static")[1]
In [6]: b.close()
In [7]: b_out.close()
In [8]: b_out=rosbag.Bag('mwe_out.bag')
In [9]: b_out_topic_info = b_out.get_type_and_topic_info("/tf_static")[1]
In [10]: b_out.close()
In [11]: print b_topic_info
{'/tf_static': TopicTuple(msg_type='tf2_msgs/TFMessage', message_count=2, connections=2, frequency=1906.5018181818182)}
In [12]: print b_out_topic_info
{'/tf_static': TopicTuple(msg_type='tf2_msgs/TFMessage', message_count=2, connections=1, frequency=1906.5018181818182)}
```
* Play back the new output bag, subscribe to `/tf_static` in a different window.
```
$ rosbag play mwe_out.bag -k
$ rostopic echo /tf_static
```
Nothing is output at all by the terminal subscriber. The only difference I can see is the connection headers being different; is there an easy way to use the bag API to fix this?

tfoote, 2019-10-16:
I'd suspect that you're not setting the connection info that marks it at latched similar to this issue: https://github.com/ros/ros_comm/pull/1372

Thanks for the detailed reproduction. However, I don't think this is a bug or feature request here so I'm going to close it here on the tracker. Please ask for more help on answers.ros.org if you need a followup question.

closed 2019-10-16 by tfoote

phil0stine, 2019-10-24:
Thanks, that did it. Changing the following 2 lines fixed it.
``` python
for topic, msg, t, cnxn_hdr in b.read_messages(return_connection_header=True):
   ...:     b_out.write(topic, msg, t, connection_header=cnxn_hdr)
```
Should I put in a PR with this feature described in the function docstrings?

tfoote, 2019-11-06:
Improvements to the clarity of the documentation is always appreciated.