ros/geometry2 #413: Issue compiling from source on kinetic. logWarn and logError

lavancig, 2019-10-03:
File has old definitions of logWarn and logError. Appended with CONSOLE_BRIDGE_ it compiles fine.

https://github.com/ros/geometry2/blob/3c51d5936805512888bd8ee2893f1de22b287829/tf2/src/buffer_core.cpp#L126

tfoote, 2019-10-15:
We're building and running CI on kinetic. Please explain your setup and why it's not working. This is patched in the melodic-devel branch which supports the newer API https://github.com/ros/geometry2/blob/1cbda1e145a1214920c9448ffd8b753bf866df67/tf2/src/buffer_core.cpp#L126

jveitchmichaelis, 2019-11-29:
If you follow the standard install for Kinetic (e.g. the install generator), it pulls the Kinetic release of tf2, but presumably a newer version of Console Bridge.

How does the CIrun it?

tfoote, 2019-12-17:
Kinetic targets Debian Jessie https://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021 so cannot move forward to the new API. See #279  for past discussion. 

These macros are updated in the melodic branch, you can use that branch. Since this works for all the targeted platforms and there's newer branches targeted at your supported platform please consider using that. We generally don't want to add things for unsupported platforms if there's easy solutions to use them via a supported method.

closed 2019-12-17 by tfoote