ros/geometry2 #417: Boost linking issue with boost 1.71 in tf2_ros

bionade24, 2019-10-11:
Hi, I'm one of the ROS melodic Archlinux maintainers, and even though `catkin 0.7.19` fixed the issue for `qt-gui-cpp`, it still appears here.
Boost version: 1.71
OS: Latest Arch clean chroot
Part: tf2_ros
Other parts built successfully
```
[ 90%] Linking CXX executable devel/lib/tf2_ros/static_transform_publisher
/usr/bin/ld: warning: libboost_thread.so.1.69.0, needed by /opt/ros/melodic/lib/libroscpp.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_chrono.so.1.69.0, needed by /opt/ros/melodic/lib/libroscpp.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_filesystem.so.1.69.0, needed by /opt/ros/melodic/lib/libroscpp.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /opt/ros/melodic/lib/libroscpp.so: undefined reference to `boost::detail::set_tss_data(void const*, boost::shared_ptr<boost::detail::tss_cleanup_function>, void*, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tf2_ros_static_transform_publisher.dir/build.make:109: devel/lib/tf2_ros/static_transform_publisher] Error 1
make[1]: *** [CMakeFiles/Makefile2:612: CMakeFiles/tf2_ros_static_transform_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable devel/lib/tf2_ros/buffer_server
/usr/bin/ld: warning: libboost_thread.so.1.69.0, needed by /opt/ros/melodic/lib/libactionlib.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_chrono.so.1.69.0, needed by /opt/ros/melodic/lib/libroscpp.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_filesystem.so.1.69.0, needed by /opt/ros/melodic/lib/libroscpp.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /opt/ros/melodic/lib/libroscpp.so: undefined reference to `boost::detail::set_tss_data(void const*, boost::shared_ptr<boost::detail::tss_cleanup_function>, void*, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tf2_ros_buffer_server.dir/build.make:110: devel/lib/tf2_ros/buffer_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:1814: CMakeFiles/tf2_ros_buffer_server.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

tfoote, 2019-10-15:
Are you sure that you have a clean environment? It looks like your copy of libroscpp.so is looking for an older version of boost, which suggests that it was built against an older version of boost. (1.69) 

And then since you're using a newer version of boost the symbols from the older version are missing.

tfoote, 2019-10-15:
Since this is being tracked at https://github.com/ros-melodic-arch/ros-melodic-tf2-ros/issues/2 I'm going to close this. It looks just to be a rebuild is necessary for the changed boost version.

closed 2019-10-15 by tfoote

bionade24, 2019-10-16:
Sorry, simply was busy and forgot to close it. It's already solved, simply needed rebuilding everything boost depending.