ros/geometry2 #424: Can't import tf2_geometry_msgs

mphamhung, 2019-10-17:
I'm using Ubuntu 18.04 Melodic
I can import tf, but not PyKDL I built with

virtualenv -p /usr/bin/python3 venv
source venv/bin/activate
catkin build --cmake-args -DPYTHON_EXECUTABLE=/path/to/virtualenv/python3.6

>>> import PyKDL
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_PyKDL)

mphamhung, 2019-10-17:
I can import PyKDL no problem in the Python2 interpreter

mairasaboia, 2019-12-17:
Any solution? Having the same problem..

tfoote, 2020-01-09:
You need to provide enough information to reproduce your problem. If you're using a virtualenv you need to make sure to compile and install pykdl into your virtualenv. Likely for python2 you're getting the library from the rosdistro or system installation. Whereas it's not compiled for python3 so it won't be available on your PYTHONPATH even if you include the system libraries.

qeef, 2022-06-03:
I have the same problem on Ubuntu 22.04 + Rolling, see https://answers.ros.org/question/344496/cannot-include-tf2_geometry_msgsh/?comment=401865#post-id-401865

Because the workaround is to include `/opt/ros/...` files in `CMakeLists.txt` directly, my guess is that the problem is in packaging for the distribution.

prasuchit, 2022-06-12:
Hi,

I'm facing the same error with Noetic. Kindly help, I'm coming up against a deadline and I need this to work:

```
import tf2_geometry_msgs
  File "/opt/ros/noetic/lib/python3/dist-packages/tf2_geometry_msgs/__init__.py", line 1, in <module>
    from .tf2_geometry_msgs import *
  File "/opt/ros/noetic/lib/python3/dist-packages/tf2_geometry_msgs/tf2_geometry_msgs.py", line 31, in <module>
    import PyKDL
ModuleNotFoundError: No module named 'PyKDL'
```

prasuchit, 2024-04-24:
@Zillaynain since it has been two years, I don't quite remember all the steps I followed to fix it, but from what I remember, I had to create a catkin_isolated_ws, clone [this](https://github.com/orocos/orocos_kinematics_dynamics.git) package there and install it using catkin_make_isolated. I think I also followed some steps on their readme to ensure the paths were sourced courectly.