ros/geometry2 #480: Possible buffer overflow.

Bitfroest, 2020-08-31:
Hello @tfoote 

It would be nice if you could double check a possible buffer overflow with the latest release `0.7.3`.

## My System:

OS: `Ubuntu 20.04 Focal`
ROS: `noetic`

Latest deb package versions:
```
ros-noetic-rviz | 1.14.1-1focal.20200826.191801
ros-noetic-tf2 | 0.7.3-1focal.20200826.180741
ros-noetic-tf2-ros | 0.7.3-1focal.20200826.183648
```

## Issue:
I automatically `apt` updated my software, so also `geometry2` packages from `0.7.2` to `0.7.3`.

Now, multiple packages using sub packages of `tf2` are terminated as follows during runtime:

```
*** buffer overflow detected ***: terminated
================================================================================REQUIRED process [rviz-12] has died!
process has died [pid 141282, exit code -6, cmd /opt/ros/noetic/lib/rviz/rviz -d /home/******/rviz/***.rviz __name:=rviz __log:=/home/***/.ros/log/06ae3f90-eba8-11ea-8283-7595d98b49cf/rviz-12.log].
log file: /home/***/.ros/log/06ae3f90-eba8-11ea-8283-7595d98b49cf/rviz-12*.log
Initiating shutdown!
================================================================================
```

## Current Workaround:
I did clone `geometry2` into my workspace and used tag `0.7.2`. Packages like `rviz` using `tf2` no longer crash. I guess there might be some change from `0.7.2` to `0.7.3` which causes that crash.

Hugal31, 2020-09-01:
The culprits might be some faulty snprintf. Example: https://github.com/ros/geometry2/blob/6afd735afc3d641ec6afa762dd0922ed8b2f6e96/tf2/src/cache.cpp#L94-L98

reinzor, 2020-09-01:
We also experience this issue:

```
rosversion tf2 
0.7.3
```

reinzor, 2020-09-01:
Tried https://github.com/ros/geometry2/pull/479 ;  this seems to resolve the issue.

doisyg, 2020-09-01:
> Tried #479 ; this seems to resolve the issue.

Indeed! Solves it for us

reinzor, 2020-09-01:
> Same here ! Got the issue after the last sync (yesterday). Our whole app is crashed. How come this was released ? This should not have passed the tests

Only crashes here when I compile in `Release`. When I compile in `Debug`, I don't encounter any problems.

tfoote, 2020-09-01:
Fixed in #479 

@reinzor That's probably due to a slightly different memory layout in debug and is probably why the tests passed too.

New release prepared here: https://github.com/ros/rosdistro/pull/26407

closed 2020-09-01 by tfoote

CodeFinder2, 2020-09-03:
When will this fix be available as Ubuntu 20.04 package (ros-noetic-geometry2)?

gavanderhoorn, 2020-09-03:
Or configure the [testing repository](http://wiki.ros.org/TestingRepository).

chrisl8, 2020-09-08:
FYI: This hit me on Saturday evening (September 5th), and I spent many hours banging my head against it.

I'm not sure how to increase visibility on this, but I think we are going to see a lot of reports on this issue.

Someone added a ROS Answers post about it here: https://answers.ros.org/question/360820/buffer-overflow-at-tf-transformlistener/ so hopefully that will help people find their way to the fix faster than I found it. :-)