ros/geometry2 #498: tf2_ros dead link for Noetic's Python API

jo-jstrm, 2021-02-12:
### Problem
[http://docs.ros.org/en/noetic/api/tf2_ros/html/python](url) results in HTTP Error 404. Same link with `melodic` instead of `noetic` works.

### Impact
Cannot access current tf2_ros Python documentation for ROS Noetic.

### Proposed Solution
Add the aforementioned resource to [http://docs.ros.org/](url).

tfoote, 2021-02-18:
The builds appear to be failing due to dot not being found. 

```
# BEGIN SUBSECTION: rosdoc_lite - tf2_ros
Invoking `rosdoc_lite` for package 'tf2_ros': python3 /tmp/rosdoc_lite/scripts/rosdoc_lite /tmp/ws/src/geometry2/tf2_ros -o /tmp/generated_documentation/api_rosdoc/tf2_ros -g /tmp/generated_documentation/symbols/tf2_ros.tag -t /tmp/generated_documentation/rosdoc_tags/tf2_ros.yaml
warning: Tag 'PERL_PATH' at line 1996 of file '/tmp/tmpeqq1ivgi' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 2018 of file '/tmp/tmpeqq1ivgi' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: tag INCLUDE_PATH: include path '$INCLUDE_PATH' does not exist
warning: source $INCLUDE_PATH is not a readable file or directory... skipping.
/tmp/rosdoc_index/noetic/symbols/rospack.tag:1576: warning: Duplicate anchor codeapi found
/tmp/ws/src/geometry2/tf2_ros/doc/mainpage.dox:6: warning: multiple use of section label 'codeapi' while adding section, (first occurrence: index.html)
sh: 1: dot: not found
sh: 1: dot: not found
error: Problems running dot: exit code=127, command='dot', arguments='"/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/time__reset__test_8cpp__incl.dot" -Tpng -o "/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/time__reset__test_8cpp__incl.png"'
error: Problems running dot: exit code=127, command='dot', arguments='"/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/message__filter_8h__dep__incl.dot" -Tpng -o "/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/message__filter_8h__dep__incl.png"'
sh: 1: dot: not found
sh: 1: dot: not found
sh: 1: dot: not found
error: Problems running dot: exit code=127, command='dot', arguments='"/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/inherit_graph_8.dot" -Tpng -o "/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/inherit_graph_8.png"'
error: Problems running dot: exit code=127, command='dot', arguments='"/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/buffer__interface_8h__incl.dot" -Tpng -o "/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/buffer__interface_8h__incl.png"'
error: Problems running dot: exit code=127, command='dot', arguments='"/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/classtf2__ros_1_1Buffer__inherit__graph.dot" -Tpng -o "/tmp/generated_documentation/api_rosdoc/tf2_ros/html/c++/classtf2__ros_1_1Buffer__inherit__graph.png"'
sh: 1: dot: not found
```

This will need to be refiled upstream on the infrastructure. I'll figure out where.

MatthijsBurgh, 2021-10-05:
@tfoote, the log you have posted only shows issues for the cpp documentation, not python.

closed 2021-10-15 by tfoote

MatthijsBurgh, 2021-10-17:
@tfoote The Python documentation on noetic is still missing. That isn't solved by https://github.com/ros-infrastructure/ros_buildfarm/pull/856. That should only have solved the missing inheritance graphs in the doxygen cpp documentation. Those are still missing, because probably no new build has been triggered, because of no changes in the actual package.

To fix the python documentation, https://github.com/ros/geometry2/pull/521, should be merged.

MatthijsBurgh, 2021-11-04:
The inheritance diagrams on the CPP docs are still missing. Does it need to be triggered? Or does it require for something to be released.