ros/geometry2 #501: Error: dynamic module does not define module export function (PyInit__tf2) After compilation

yanjunyangmonash, 2021-03-08:
Hi, I'm using Ubuntu 18.04 and ROS melodic. I first got this error when I tried to import tf2_ros, and I followed this [instruction](https://answers.ros.org/question/326226/importerror-dynamic-module-does-not-define-module-export-function-pyinit__tf2/) to install the tf2_ros. Since the other packages have to be compiled using `catkin build`, I made some modifications. Below are commands I used to compile it for python3
```
cd ~/catkin_ws
source devel/setup.bash
wstool init
wstool set -y src/geometry2 --git https://github.com/ros/geometry2 -v melodic-devel
wstool up
rosdep install --from-paths src --ignore-src -y -r

catkin build --cmake-args \
            -DCMAKE_BUILD_TYPE=Release \
            -DPYTHON_EXECUTABLE=/usr/bin/python3 \
            -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m \
            -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so

source devel/setup.bash
```
After doing these, when I imported tf2_ros, I still get the same error
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/yanjun/catkin_ws/src/geometry2/tf2_ros/src/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/home/yanjun/catkin_ws/src/geometry2/tf2_py/src/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ModuleNotFoundError: No module named 'tf2_py._tf2'
```

yellow07200, 2021-04-07:
I have the exactly same issue. Have you solved this problem?

yanjunyangmonash, 2021-04-20:
> I have the exactly same issue. Have you solved this problem?

From my note
1. I created a conda virtual env in python 2.7
2. git clone https://github.com/ros/geometry2.git - b melodic-devel
3. catkin build

And it worked for me, but I'm not sure if this could help you or not

closed 2025-05-31 by sloretz