ros/geometry2 #515: Documentation on convert/transform registration in Python

MatthijsBurgh, 2021-08-08:
Is there any documentation on the convert/transform registration in Python?

MatthijsBurgh, 2021-08-20:
ping ;)

MatthijsBurgh, 2021-09-21:
@tfoote ping ;)

tfoote, 2021-09-27:
There's no significant docs. You can find examples [here](https://github.com/ros/geometry2/blob/noetic-devel/tf2_geometry_msgs/src/tf2_geometry_msgs/tf2_geometry_msgs.py) and [here](https://github.com/ros/geometry2/blob/noetic-devel/tf2_sensor_msgs/src/tf2_sensor_msgs/tf2_sensor_msgs.py)

For converting:
You write a to and from message and register them with `tf2_ros.ConvertRegistration()`

For transforming: 
You write the function to add it and then register it using the `tf2_ros.TransformRegistration()`

MatthijsBurgh, 2021-09-29:
For me it is unclear why all the `to_msg` and `from_msg` functions can just return the same message.

I would expect that if I have class A and I want to transform to class B, but there is no direct conversion from A to B, it will use the `to_msg` and `from_msg` way. So I have to go from A to a msg. Lets call it X. And then convert msg X to class B.

In my mind, msg X has to be ONE type of msg. As a random pair of `to_msg` and `from_msg` aren't compatible, unless the `to_msg` functions outputs a type, which the `from_msg` functions accepts. Otherwise AttributeError are doomed to happen IMO.

Maybe I am missing something, please explain.

tfoote, 2022-02-05:
1) The registrations are keyed on the datatype. A `Vector3Stamped` converter and a `PoseStamped` converter will never be chained. 

2) Those the`to_msg` and `from_msg` will not be paired like you said because they have different datatype keys.

The noop returns are required because in python we can't detect what the datatype is so we don't know if the datatype matches the input datatype. So `convert` always triggers `to_msg` then `from_msg` to make the conversion. If either end of that goal is already a message it will be the noop.