ros/geometry2 #520: Release 0.6.7 never made it into Melodic distro?

rhaschke, 2021-09-24:
@tfoote, I'm wondering why the release 0.6.7 didn't actually make it into the Melodic distro? Did you just forget to bloom?
I remember that we had issues with 0.6.6, finally reverting it: https://github.com/ros/rosdistro/pull/23469.
But meanwhile, the offending commit was reverted (3927d64c49ddc9f125bbf990e54f52ac981b9a42) and new ones included instead (#444). Ah, I just noticed that we didn't have a release since #444 was merged. Maybe that's the reason?

tfoote, 2022-01-05:
Yeah, all the primary changes have all been for windows. And with the release and regression I held back on 0.6.7 because of the identified issues that we fixed later.

For the windows use cases it's enough currently to be on the development tip. So I haven't taken time to fully QA everything for another release. To make sure we don't have another release that we break the linux build trying to resolve the windows build.

gavanderhoorn, 2022-01-14:
Having #377 in a release would be nice though.

Just ran into a situation where not seeing `--precision` caused some confusion, seeing as the Noetic system I was using did have it, and according to the changelog it was in Melodic as well (status page showed it was never released though).

closed 2025-05-31 by sloretz