ros/geometry2 #523: PermissionError when using view_frames

roversch, 2021-11-19:
I'm on ROS `noetic`, installed `tf2_tools` through `apt`.

I start a Gazebo simulation and simply run `rosrun tf2_tools view_frames.py`. I get 

```
[INFO] [1637331888.797080, 0.000000]: Listening to tf data during 5 seconds...
[INFO] [1637331894.153208, 234.577000]: Generating graph in frames.pdf file...
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/tf2_tools/view_frames.py", line 81, in <module>
    main()
  File "/opt/ros/noetic/lib/tf2_tools/view_frames.py", line 52, in main
    with open('frames.gv', 'w') as f:
PermissionError: [Errno 13] Permission denied: 'frames.gv'
```

roversch, 2021-11-19:
Actually, this only happens when I run it from a folder that I don't have permission to write in.

If I start from e.g. my home folder, everything works as expected.

tfoote, 2022-02-04:
Yeah, you need permissions to write to file in your current directory.

closed 2022-02-04 by tfoote