ros/geometry2 #536: New release for Noetic

MatthijsBurgh, 2022-05-23:
Sadly no new release was published before the recent Noetic sync.

I think it would be really appreciated by the community when a new release would be published before the next Noetic sync.

MatthijsBurgh, 2022-06-17:
@tfoote friendly ping

MatthijsBurgh, 2022-07-27:
@tfoote ping ping ;)

MatthijsBurgh, 2022-08-25:
@tfoote ping ping ping ;)

MatthijsBurgh, 2022-09-01:
@tfoote ping ping ping ping ;)

MatthijsBurgh, 2022-09-16:
@tfoote ping ping ping ping ping ping ;)

gavanderhoorn, 2022-09-17:
Would it be an idea to add some more people to the release/maintenance team to not have to bother @tfoote with requests like these?

MatthijsBurgh, 2022-09-17:
@gavanderhoorn sounds like a plan. I have no experience with releasing ROS packages. So would you be open to it?

gavanderhoorn, 2022-09-17:
I expect @tfoote to respond with "it's much more than just running Bloom: checking PRs, running tests, making sure there are no regressions, keeping track of the direction of development, etc, etc".

I'd not be unwilling to contribute some time to "just run Bloom", but I don't want to become just-another-bottleneck for `geometry2` releases / maintenance.

For such a core package though, it would seem prudent to make sure there are sufficient maintainers to at least release things every-now-and-then.

MatthijsBurgh, 2022-10-06:
@tfoote can we get a new release in before the next sync? https://discourse.ros.org/t/noetic-sync-before-roscon-iros/27646

tfoote, 2022-10-07:
Sorry for the delay , I will review what's in the branch is releasable and get it out shortly.

tfoote, 2022-10-11:
Sorry I don't want to push it in right before a sync happens to make sure that we have a good amount of testing it will land right after the sync. https://github.com/ros/rosdistro/pull/34896

However you can test it in the testing/shadow-fixed repo shortly thereafter

closed 2022-10-11 by tfoote