ros/geometry2 #541: dotransform TransformStamped datatype not process child_frame_id

darkchooselight, 2022-09-15:
the code do not process child_frame_id transform. look below:
tf2_geometry_msgs/tf2_geometry_msgs.h

```
template <>
inline
void doTransform(const geometry_msgs::TransformStamped& t_in, geometry_msgs::TransformStamped& t_out, const geometry_msgs::TransformStamped& transform)
  {
    tf2::Transform input;
    fromMsg(t_in.transform, input);

    tf2::Transform t;
    fromMsg(transform.transform, t);
    tf2::Transform v_out = t * input;

    t_out.transform = toMsg(v_out);
    t_out.header.stamp = transform.header.stamp;
    t_out.header.frame_id = transform.header.frame_id;
  }
```

tfoote, 2022-10-06:
That does look like it's missing the logic to transfer the `child_frame_id` if t_in and t_out aren't the same.

tfoote, 2023-09-22:
I was reviewing the PR #555 proposed to resolve this and I think this behavior is actually correct. See my logic there.

closed 2023-09-22 by tfoote